espressif/esp-modbus #108: Send request not updating buffer

RoootNoodle, 2025-03-27:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

I want to read a 32bit float holding register value using the library as a master. As shown in the images, the correct value is received from the slave but does not get loaded into the buffer. I am not sure whether there is a bug on the library side or if made a mistake in the configuration. Please note the the guide provided by the link is very outdated and not all the functions shown in the link is available in the latest release.

https://docs.espressif.com/projects/esp-modbus/en/stable/esp32/overview_messaging_and_mapping.html

The image below show the code sent to the slave: 00 03 00 00 00 02 c5 da
With a reply: 00 03 04 41 45 70 a4 cb 61
The reply should be decoded as 12.34 but as seen the Float value = 0

[image]

The loop to continuously send a request

[image]

The setup code

[image]

I also tried implementing a descriptor, but the result was the same, with only a 0 on return. As per my understanding, the mbc_master_send_request() function should trigger a callback function mbc_reg_holding_master_cb() to update the buffer value, which does not happen. The mbc_reg_holding_master_cb() is defined as a weak function, which im not certain whether to include some reference to it in my main.c.

I could not find help or a guide on the internet, any advice would be appreciated

alisitsyn, 2025-03-28:
Hello @RoootNoodle,

Thank you for your report. I can see in your log the response has been correctly received, but the data is incorrect.

> I also tried implementing a descriptor, but the result was the same, with only a 0 on return. As per my understanding, the mbc_master_send_request() function should trigger a callback function [...]

This issue can be due to incorrect linking of the `mbc_reg_holding_master_cb` in the library (`mbm_controller_iface->mb_base->rw_cbs.mbc_reg_holding_master_cb`). In normal use case you don't need to reference the callback in your code. The issue will be checked accordingly.

Could you let me know some aspects:
1. Which version of esp-idf you are using? 
2. Please send me the `your_project_name.map` file from your build folder?
3. Can you repeat this issue with other readings of parameters (input registers for example)? 

 Thanks.

alisitsyn, 2025-03-28:
Thank you for the update. I will take a look to it and investigate when possible.

> Should I make use of the mb_parameter_descriptor if the data is being read with mbc_master_send_request()?

It actually does not matter if you setup reading using the data dictionary or send the request directly. The mapping rw callback function shall be executed from the command handler from the event task context in both cases.

RoootNoodle, 2025-03-28:
I am using ESP-IDF v5.4.0, I used the same code as provided but the data still did not map.

I will move the library and try again.

RoootNoodle, 2025-03-28:
[image]

I added the first two lines, there is still no change in the output. Could the problem with the setup in my code or is this library orientated?

I'm attaching my main.c

[main.zip](https://github.com/user-attachments/files/19508521/main.zip)

alisitsyn, 2025-03-28:
Please to remove the weak attribute in the function prototype to check it instantiation: 
esp_modbus_master.h:
```
mb_err_enum_t mbc_reg_holding_master_cb(mb_base_t *inst, uint8_t *reg_buffer, uint16_t address, uint16_t n_regs, mb_reg_mode_enum_t mode) __attribute__ ((weak)); //  >>>>>>>>> please remove `__attribute__ ((weak))` here
```
then check if the function is called on request.

RoootNoodle, 2025-03-29:
The changes have been made, there is still not change. The callback is not being executed.

[image]

The output after removing weak attribute and adding logs inside the callback.

Should the library be moved to components folder every time or is this done only for testing?

alisitsyn, 2025-03-31:
> Should the library be moved to components folder every time or is this done only for testing?

This is just to check the reason on your side.
Please let me know which chip (module) is used on your side. Send me whole log of the communication instead of picture. Thanks.

RoootNoodle, 2025-04-01:
I got it working, thank you very much. The problem was with broadcasting the request to all the slaves and not requesting data from an individual slave. Changing the req.slave_addr = 1 fixed the problem.

closed 2025-04-01 by RoootNoodle