espressif/esp-modbus #110: Modbus Slave TCP - Unable to open more than 5 sockets in a row

IGD-james, 2025-04-02:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

Opening and closing a TCP socket to the Modbus slave should work continuously.

### Actual behavior (suspected bug)

After opening 5 sockets, it then fails in mb_drv_open.

``pdrv_ctx->mb_node_open_count`` is incremented for each opened connection. But it is never decremented anywhere, not that I can find anyway. Therefore the below check fails, resulting in errors when trying to open new Modbus clients.
```
if (pdrv_ctx->mb_node_open_count > MB_MAX_FDS) {
    goto err;
}
```

I added a custom message "Too many Modbus nodes open" to show this. The log below shows this condition.

### Error logs or terminal output

```shell
D (81254) mb_port.tcp.slave: loop:0x3fcbb1f0  mbs_on_timeout: fd: -1
D (81346) mb_driver: 0x3fcbb1f0, listen_sock is active.
I (81346) port.utils: Socket (#60), accept client connection from address[port]: 192.168.4.2[55907]
D (81350) mb_driver: port_accept_connection: 60
D (81354) mb_driver: 0x3fcbb1f0, open vfd: 0, sl_addr: 00, node: 192.168.4.2:55907
D (81358) mb_driver: mb_drv_open: addr_info.fd=60, fd=0
E (81366) mb_driver: Too many Modbus nodes open!
E (81370) mb_driver: 0x3fcbb1f0, unable to open node: 192.168.4.2
D (81578) mb_driver: 0x3fcbb1f0, fd event get: 0x100:-1, |MB_EVENT_TIMEOUT
D (81578) event: running post loop:0x3fcbb1f0:256 with handler 0x4200e698 and context 0x3fcbd064 on loop 0x3fcbb5dc
--- 0x4200e698: mbs_on_timeout at C:/Users/James/Documents/Dev/basic/managed_components/espressif__esp-modbus/modbus/mb_ports/tcp/port_tcp_slave.c:592
```

### Steps to reproduce the behavior

TCP example slave.
2. Connect and reconnect using Modbus client like Simply Modbus.
3. Perform this multiple times in a row and eventually the client won't be able to connect.

### Project release version

latest

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

Windows 10

### Shell

ZSH

### Additional context

Using ESP-IDF V5.4.0 with WiFi AP setup. ESP32-S3

IGD-james, 2025-04-02:
There is also a resource utisilisation problem where, if the function ``mb_drv_open`` does fail. It will still have called ``init_queues`` which allocates some resources, but in the error handler these are not deallocated.

This just stops all connections dead once all queues are used up.

I found changing the error handler to this helped. There may be a nicer way to do this...
```
err:
    if (pnode_info->rx_queue || pnode_info->tx_queue) {
        delete_queues(pnode_info);
    }
    free(pnode_info);
    pdrv_ctx->mb_nodes[fd] = NULL;
    mb_drv_unlock(ctx);
    return UNDEF_FD;
```

bugrayanik, 2025-04-03:
[image]

There is a Macro which is set to 5 for max TCP connections, you can change it through menuconfig or from sdkconfig.
Let me know if that solves it.

alisitsyn, 2025-04-03:
Hi @IGD-james

Thank you for reporting. 
The disconnection of slaves are supposed to be handled in `mbs_on_error` handler. The destroy of connections and associated queues are performed in `mb_drv_close()`.
This is known issue but fix was not completed. There are some additional aspects related to connection management in slave. This will be handled accordingly. I will try to check again and fix when possible.

alisitsyn, 2025-04-22:
@klew , 

Thanks for your report. I am pretty busy with other things now. I aware of this and will try to realize simple fix for this ASAP in new release.

iotsystem, 2025-05-22:
I have also encountered this problem, when can it be fixed?

alisitsyn, 2025-05-27:
@iotsystem, 

Thank you for the information. The fix is available and goes through the formal process. It is planned to be merged in June. Sorry for the delays and possible inconvenience.

dsyx, 2025-06-04:
I had the same problem:

```
[17:35:13.419]收←◆I (998206) port.utils: Socket (#60), accept client connection from address[port]: 10.10.10.10[64015]
E (998206) mb_driver: 0x4084505c, unable to open node: 10.10.10.10

[17:35:14.730]收←◆I (999516) port.utils: Socket (#61), accept client connection from address[port]: 10.10.10.10[64016]
E (999516) mb_driver: 0x4084505c, unable to open node: 10.10.10.10

[17:35:16.053]收←◆I (1000836) port.utils: Socket (#62), accept client connection from address[port]: 10.10.10.10[64017]
E (1000836) mb_driver: 0x4084505c, unable to open node: 10.10.10.10

[17:35:17.150]收←◆I (1001936) port.utils: Socket (#63), accept client connection from address[port]: 10.10.10.10[64018]
E (1001936) mb_driver: 0x4084505c, unable to open node: 10.10.10.10

[17:35:18.585]收←◆E (1003366) port.utils: Unable to accept connection: errno=23
E (1003366) mb_driver: 0x4084505c, unable to open node: 10.10.10.10

[17:35:22.539]收←◆E (1007316) port.utils: Unable to accept connection: errno=23
E (1007326) mb_driver: 0x4084505c, unable to open node: 10.10.10.10
```

In addition, I found that tcp.slave could not handle requests correctly when sending them at a 200ms period:

```
[17:22:40.401]收←◆I (245186) port.utils: Socket (#55), accept client connection from address[port]: 10.10.10.10[63603]

[17:22:46.644]收←◆E (251436) mb_port.tcp.slave: queue can not find the item to send.
E (251436) mb_port.tcp.slave: incorrect frame to send.
E (251436) mb_object.slave: 0x40843634:frame send error. 5

[17:22:47.743]收←◆E (252536) mb_port.tcp.slave: queue can not find the item to send.
E (252536) mb_port.tcp.slave: incorrect frame to send.
E (252536) mb_object.slave: 0x40843634:frame send error. 5

[17:22:48.793]收←◆E (253586) mb_port.tcp.slave: queue can not find the item to send.
E (253586) mb_port.tcp.slave: incorrect frame to send.
E (253586) mb_object.slave: 0x40843634:frame send error. 5

[17:22:49.843]收←◆E (254636) mb_port.tcp.slave: queue can not find the item to send.
E (254636) mb_port.tcp.slave: incorrect frame to send.
E (254636) mb_object.slave: 0x40843634:frame send error. 5
```

dsyx, 2025-06-20:
Hi @alisitsyn ,
I noticed the recently released v2.1.0 and updated to it.
However, after testing, I found that the problem still exists. Under `CONFIG_FMB_TCP_PORT_MAX_CONN=2`, the third connection will cause the application to crash and reset, as shown in the following log:
```
[11:27:25.817]收←◆I (5537) port.utils: mdns delegate hostname set to: [mb_slave_tcp_01]
I (5537) port.utils: IP: 10.10.49.115
I (5537) port.utils: GW: 10.10.0.1
I (5547) port.utils: NETMASK: 255.255.0.0
I (5547) port.utils: Socket (#54), listener  on port: 502, errno=0
I (5547) mb_port.tcp.slave: loop:0x40844e94  mbs_on_ready: fd: -1, bind is done

[11:28:28.896]收←◆I (68607) port.utils: Socket (#55), accept client connection from address[port]: 10.10.10.10[64500]

[11:28:58.858]收←◆I (98577) port.utils: Socket (#56), accept client connection from address[port]: 10.10.10.10[64520]

[11:29:11.159]收←◆I (110877) port.utils: Socket (#57), accept client connection from address[port]: 10.10.10.10[64521]
E (110877) mb_driver: 0x40844e94, unable to open node: 10.10.10.10
Guru Meditation Error: Core  0 panic'ed (Cache error). 
Cache access error

Core  0 register dump:
MEPC    : 0x4205b4ce  RA      : 0x42059974  SP      : 0x408463b0  GP      : 0x40822094  
TP      : 0x408464d0  T0      : 0x40022494  T1      : 0xffffffe0  T2      : 0xffffffff  
S0/FP   : 0x40844e94  S1      : 0x4082f4bc  A0      : 0x40844e94  A1      : 0x4082f4c0  
A2      : 0x00000032  A3      : 0x4082f4bc  A4      : 0x408466f0  A5      : 0x428f7e58  
A6      : 0x4205994e  A7      : 0x42079d3a  S2      : 0x00000000  S3      : 0x00002b21  
S4      : 0x00000005  S5      : 0x00000000  S6      : 0x00000005  S7      : 0x00000000  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x00000000  T4      : 0x00000000  T5      : 0x00000000  T6      : 0x00000000  
MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000019  MTVAL   : 0x00001141  
MHARTID : 0x00000000  

Stack memory:
408463b0: 0x00000000 0x408450fc 0x00000000 0x420fd4b4 0x00000000 0x408450fc 0x00000000 0x420fde00
408463d0: 0x00000101 0x40845448 0x00000040 0x4082f4bc 0x00000101 0x40845448 0x00000040 0x4082f4bc
408463f0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40846410: 0x00000000 0x40844e94 0x00000000 0x4205b29e 0x00000080 0x00000039 0x4083a98c 0x4083a98c
40846430: 0x00000001 0xfc090000 0x00000002 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40846450: 0x00000002 0x00000001 0x00000040 0x01800000 0x00000000 0x00000000 0x00000080 0x4082d494
40846470: 0x4083a98c 0x4083a98c 0x00000001 0xfc090000 0x00000002 0x00000000 0x00000000 0x00000000
40846490: 0x00000000 0x00000000 0x00000000 0x40810442 0x00000000 0x00000000 0x00000000 0x00000000
408464b0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
408464d0: 0xa5a5a5a5 0x00000150 0x40846220 0x00002b20 0x40825e4c 0x40825e4c 0x408464d8 0x40825e44
408464f0: 0x0000000f 0x40832580 0x40832580 0x408464d8 0x00000000 0x0000000a 0x408454d4 0x645f626d
40846510: 0x745f7672 0x745f7063 0x006b7361 0x408464d0 0x0000000a 0x00000001 0x40846f04 0x42008852
40846530: 0x00000080 0x4082dda8 0x4082de10 0x4082de78 0x00000000 0x00000000 0x00000001 0x00000000
40846550: 0x00000000 0x00000000 0x4200820e 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40846570: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40846590: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
408465b0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
408465d0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
408465f0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40846610: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000010 0x40845448
40846630: 0x00000000 0x40846640 0x00000000 0x0000000c 0x00000001 0x40846650 0x40846670 0x0000000c
40846650: 0x40846660 0x00000000 0x00000000 0x0000000c 0x42059356 0x40844e94 0x40846658 0x0000000c
40846670: 0x00000002 0x40846680 0x408466a0 0x0000000c 0x40846690 0x00000000 0x00000000 0x0000000c
40846690: 0x42058d96 0x40844e94 0x40846688 0x0000000c 0x00000008 0x408466b0 0x408466d0 0x0000000c
408466b0: 0x408466c0 0x00000000 0x00000000 0x0000000c 0x420594bc 0x40844e94 0x408466b8 0x0000000c
408466d0: 0x00000040 0x408466e0 0x40846700 0x0000000c 0x408466f0 0x00000000 0x00000000 0x0000000c
408466f0: 0x4205994e 0x40844e94 0x408466e8 0x0000000c 0x00000010 0x40846710 0x40846730 0x0000000c
40846710: 0x40846720 0x00000000 0x00000000 0x0000000c 0x42059704 0x40844e94 0x40846718 0x0000000c
40846730: 0x00000020 0x40846740 0x40846760 0x0000000c 0x40846750 0x00000000 0x00000000 0x0000000c
40846750: 0x4205955a 0x40844e94 0x40846748 0x0000000c 0x00000080 0x40846770 0x40846790 0x0000000c
40846770: 0x40846780 0x00000000 0x00000000 0x0000000c 0x420599e4 0x40844e94 0x40846778 0x0000000c
40846790: 0x00000100 0x408467a0 0x00000000 0x0000000c 0x408467b0 0x00000000 0x00000000 0x0000000c

ELF file SHA256: de96cf623

Rebooting...
```

And there is a chance to trigger the following infinite loop reset:

```
E (90192) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (90192) task_wdt:  - IDLE (CPU 0)
E (90192) task_wdt: Tasks currently running:
E (90192) task_wdt: CPU 0: mb_drv_tcp_task
[... cut ...]
A2      : 0x40846470  A3      : 0x40846418  A4      : 0x40844eec  A5      : 0x00000000  
A6      : 0x00000001  A7      : 0x00000001  S2      : 0x00000000  S3      : 0x00000000  
S4      : 0x00000000  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x00000000  T4      : 0x00000000  T5      : 0x0ccccccc  T6      : 0x00000019  
MSTATUS : 0x00001889  MTVEC   : 0x40800001  MCAUSE  : 0xdeadc0de  MTVAL   : 0xdeadc0de  
MHARTID : 0x00000000
```

dsyx, 2025-06-24:
Hi @alisitsyn ,

I have created a [new issue](https://github.com/espressif/esp-modbus/issues/127), please go to the new issue for discussion.

Thanks.

alisitsyn, 2025-06-25:
@dsyx ,

> In addition, I found that tcp.slave could not handle requests correctly when sending them at a 200ms period:

if you still have this bug, could you open a separate issue for this as well. This will allow to handle the issues better and I will check it as soon as possible. 

Thanks.

dsyx, 2025-06-26:
Hi @alisitsyn ,

> [@dsyx](https://github.com/dsyx) ,
>
>
> if you still have this bug, could you open a separate issue for this as well. This will allow to handle the issues better and I will check it as soon as [...]

An issue has been created for this issue, we can move the discussion there.

Thanks

alisitsyn, 2025-08-26:
Fixed in main with commit [7b2dd251].