espressif/esp-modbus #111: Modbus Slave RTU - Master receives invalid response from ESP32-C3 Slave

Ruanfc, 2025-04-05:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

Downloaded mb_serial_master and mb_serial_slave examples from main branch. The arrangement is as follows:

- ESP32 Master + MAX3485 chip module;
- ESP32 Slave + MAX485 chip module        (MODBUS ADDR = 1)
- ESP32-C3 Slave + MAX3485 chip module  (MODBUS ADDR = 2)

I expected to run the examples with no log errors.

This is the MAX3485 module:
[image]

And this is the MAX485 module:
[image]

### Actual behavior (suspected bug)

When I try running Master with MB_DEVICE_ADDR =1 to the first slave, it **works perfectly** even when slave 2 is wired to the RS485 net.
However, if I replace **MB_DEVICE_ADDR = 2** without changing anything to the circuit, from the console one can see that the slave is indeed receiving the message, but in the master log we always receive error 0x108 (ESP_ERR_INVALID_RESPONSE).

### Error logs or terminal output

```shell
`E (29646) MASTER_TEST: Characteristic #8 (RelayP2) read fail, err = 0x108 (ESP_ERR_INVALID_RESPONSE).`
```

### Steps to reproduce the behavior

1. Downloaded both master and slave example from main branch;
2. Unzipped the examples and fixed the path to `mb_example_common` in `idf_component.yml`;
3. added .vscode configuration folder and `idf.py reconfigure`
4. Menuconfig on both examples to set UART, baudrate, RX, TX, RTS pins and RTU mode
5. Build, flash and monitor on each slave and master.

### Project release version

v2.0.2

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

Windows 11

### Shell

PowerShell

### Additional context

Master UART pins are:
  - RX: GPIO16, 
  - TX: GPIO17,
  - RTS: GPIO18,
  - binded to UART2
ESP32 Slave 1 UART pins are the same as master;
ESP32-C3 Slave 2 UART pins are:
 - RX: GPIO08,
 - TX: GPIO09,
 - RTS: GPIO10,
 - binded to UART1 (as there is no UART2 on ESP32-C3)
 (But already tested RX: GPIO06, TX: GPIO07, RTS: GPIO04)

I'm just wondering if there is a kind of conflict on UART1 when running the example as slave on ESP32-C3. But I don't know to ensure this is not the problem.

I don't think the problem is the example code itself since I already tested the slave example in the espressif/esp-modbus:^1.0.

Ruanfc, 2025-04-05:
Sorry for my English. Past tense of verb to bind is bound.

Ruanfc, 2025-04-10:
To answer your questions:
1.  No, none of them have PSRAM chip integrated;
2.  I didn't change anything to the examples codes, so at first the master didn't find any information about the slave and them any any characteristic could be read giving that 0x108 error.
3.  Indeed! You're absolutely right and I didn't think about it. So I manage to replace it so each of them are only MAX-3485;
4.  I couldn't record as I don't have a logic analyzer and I don't know any 3.3V RS-485 sniffer project as well.
---
Either way... I managed to execute new tests.

### First test
In the first test the circuit is as follows:
* ESP32 + MAX-3485 on slave;
* ESP32-C3 + MAX-3485 on master;
In this test I realized two things: 1) I didn't realized the **FMB_EXT_TYPE_SUPPORT** bits were different on slave if compared to the master (As you said, I had *to make sure the register areas are configured correctly*); 2) I've found a lil poor wiring conection on the RS-485 socket in the master and then I managed to correct it. 

### Second test
The second test was adding a second slave for testing communication:
* ESP32-C3 + MAX-3485 on slave 1
* ESP32  + MAX-3485 on slave 2
* ESP32-C3 + MAX-3485 on master
Compiled and run the master with `MB_DEVICE_ADDR1 = 1` and ran again with `MB_DEVICE_ADDR1 = 2`. On both attempts I had **SUCCESS**.  The log was printed in the file as follows:

[esp-log.txt](https://github.com/user-attachments/files/19693591/esp-log.txt)

TL;DR:
You guessed it right. There were to problems I managed to correct:
1) Don't mix MAX-485 with MAX-3485 chips modules in the same RS-485 net since there is voltage incompatibility;
2) I didn't pay attention to **FMB_EXT_TYPE_SUPPORT** bit since it may cause the register areas to be configured different on master regarding the slave.

Thank you for your support!

alisitsyn, 2025-04-11:
Thank you for your feedback. Good luck!