espressif/esp-modbus #113: After updating to version 1.0.17 the project stopped building

kotyara12, 2025-04-23:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

I have a greenhouse project with many RS485/modbus sensors.
The project was created in PlatformIO - this is important!
Today I updated ESP-IDF to framework-espidf @ 3.50400.0 (5.4.0), and esp-modbus was updated from 1.0.13 to 1.0.17, and the project stopped building, although everything compiles!

### Actual behavior (suspected bug)

```
Archiving .pio\build\esp32dev\libea2\libsensors.a
Linking .pio\build\esp32dev\firmware.elf
C:/Users/kotyara12/.platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld.exe: .pio/build/esp32dev/esp-idf/espressif__esp-modbus/libespressif__esp-modbus.a(mbc_serial_master.c.o):(.literal.mbc_serial_master_send_request+0x14): undefined reference to `eMBMasterReqReportSlaveID'
C:/Users/kotyara12/.platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld.exe: .pio/build/esp32dev/esp-idf/espressif__esp-modbus/libespressif__esp-modbus.a(mbc_serial_master.c.o): in function `mbc_serial_master_send_request':
C:\Projects\PlatformIO\village_greenhouse/managed_components/espressif__esp-modbus/freemodbus/serial_master/modbus_controller/mbc_serial_master.c:193:(.text.mbc_serial_master_send_request+0x128): undefined reference to `eMBMasterReqReportSlaveID'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32dev\firmware.elf] Error 1
```

### Error logs or terminal output

```shell

```

### Steps to reproduce the behavior

Executing task in folder village_greenhouse: C:\Users\kotyara12\.platformio\penv\Scripts\platformio.exe run 

Processing esp32dev (platform: espressif32; framework: espidf; board: esp32dev)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (6.10.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-espidf @ 3.50400.0 (5.4.0)
 - tool-cmake @ 3.16.4
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - tool-idf @ 1.0.1
 - tool-mconf @ 1.4060000.20190628 (406.0.0)
 - tool-ninja @ 1.9.0
 - tool-riscv32-esp-elf-gdb @ 11.2.0+20220823 
 - tool-xtensa-esp-elf-gdb @ 11.2.0+20230208
 - toolchain-esp32ulp @ 1.23800.240113 (2.38.0)
 - toolchain-xtensa-esp-elf @ 14.2.0+20241119
Reading CMake configuration...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain+, Compatibility ~ soft
Found 42 compatible libraries

### Project release version

1.0.17

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

Windows 10

### Shell

PowerShell

alisitsyn, 2025-04-24:
@kotyara12 ,

I can not check the compilation under platformio now. From the component standpoint for v1.0.17 everything is correct and the prototypes of `eMBMasterReqReportSlaveID` are defined correctly. All tests are passed. I can recommend you trying  to clean the project and component and rebuild again.

kotyara12, 2025-04-24:
Of course, I cleaned the project. And not just once.
I return version 1.0.13 - the project builds successfully. I change it to 1.0.17 - the linker does not find the function.
Perhaps this is because the project is incorrect?
What is characteristic is that the problem is not only with me. I started checking this after one of my blog subscribers complained about the problem.

As for me, 80% of the code in this library can simply be thrown away. There is no need for all these registrations, which no one wants to deal with and waste time. All my projects with modbus are great, the most complex ones, with dozens of devices on the bus, are managed with a single function mbc_serial_master_send_request().

alisitsyn, 2025-04-24:
> Maybe this is because the project is incorrect?
What is typical is that I am not the only one with this problem. I started checking this after one of my blog subscribers complained about the problem.

My fail, I forget to check if the slave id support is enabled. The `CONFIG_FMB_CONTROLLER_SLAVE_ID_SUPPORT` is enabled by default but I guess in your sdkconfig it is not set. This should  a reason.

alisitsyn, 2025-04-24:
> And what does SLAVE_ID have to do with anything??? As far as I understand, it's from a completely different opera

The Slave ID support feature has been updated in v1.0.17 and it causes the issue you described when the otion above is disabled as in your case.

> And in the second version, as far as I was told, it is generally impossible to use mbc_serial_master_send_request() directly, without initializing these terribly ugly structures of yours

https://docs.espressif.com/projects/esp-modbus/en/stable/esp32/applications_and_references.html#_CPPv423mbc_master_send_requestPvP18mb_param_request_tPv

It is up to you what to use in your perfect software.

kotyara12, 2025-04-24:
Okay, I'll roll back to 1.0.13 and that's the end of it

closed 2025-04-24 by kotyara12

alisitsyn, 2025-04-25:
Merged to master with commit a8c7c36fb907ba093cf1001fb8ea7a43d57f4d13