espressif/esp-modbus #118: Both V1 and V2 are extremely difficult to use.

lbxhacker, 2025-05-07:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

 I have been using this library file to continuously modify and adjust for five days, but there are still inexplicable memory crash issues. Especially when initializing the TCP master station, even after deleting and releasing resources, it still cannot be initialized normally. To be honest, this library is very difficult to use. Another point I want to joke about is the TCP case. Why not be direct Can it be implemented using a normal expression method, such as char* dest_ip="192.168.1.100"? Rather, is it to be achieved through the so-called command receiving method? Is this to show off skills? My mood has become very bad because of writing adaptation for this library

lbxhacker, 2025-05-07:
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40382c8d  PS      : 0x00060733  A0      : 0x8202f49c  A1      : 0x3fcd0190
--- 0x40382c8d: xEventGroupClearBits at C:/esp/ESP-IDF-5.3.2/esp-idf/components/freertos/FreeRTOS-Kernel/event_groups.c:504

A2      : 0x00000000  A3      : 0x00000080  A4      : 0x8038269e  A5      : 0x3fc9aaa0
A6      : 0x0000001c  A7      : 0x00000000  A8      : 0x80382c8b  A9      : 0x3fcd0170
A10     : 0x00000001  A11     : 0xffffffff  A12     : 0x00060720  A13     : 0x00000002
A14     : 0xffffffff  A15     : 0x3c2f0aac  SAR     : 0x00000019  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xfffffffc

Backtrace: 0x40382c8a:0x3fcd0190 0x4202f499:0x3fcd01b0 0x4202ce3d:0x3fcd01e0 0x420179bf:0x3fcd0210
--- 0x40382c8a: vPortEnterCritical at C:/esp/ESP-IDF-5.3.2/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:567
---  (inlined by) xEventGroupClearBits at C:/esp/ESP-IDF-5.3.2/esp-idf/components/freertos/FreeRTOS-Kernel/event_groups.c:498
--- 0x4202f499: mbc_tcp_master_stop at C:/Project/GBTX-PRO/components/esp-modbus-2.0.2/modbus/mb_controller/tcp/mbc_tcp_master.c:114
--- 0x4202ce3d: mbc_master_stop at C:/Project/GBTX-PRO/components/esp-modbus-2.0.2/modbus/mb_controller/common/esp_modbus_master.c:221
--- 0x420179bf: Modbus_Master_TCP_init at C:/Project/GBTX-PRO/components/Cat-Modbus/Cat-Modbus_Master.h:57
---  (inlined by) modbus_tcp_creat_task at C:/Project/GBTX-PRO/src/Form/Form1_Modbus_Master.h:837

**This is a crash that occurs after creating TCP communication. It doesn't happen all the time; it always crashes unexpectedly**

Godoy-Octavio, 2025-05-09:
Hello, @lbxhacker 

Thank you for reporting this issue. In order to reproduce this error we need more information, could you provide the initialization sequence code you are using?

We are aware that the modbus example code is not that straightforward, we are currently working on a simplified version of it.

Please refer to this alternative modbus example with a simpler approach:

[Simplified Modbus Master example](https://github.com/alisitsyn/modbus_support/tree/mb_support/simple_modbus_master)
	
Let us know if you encounter any further issues!

alisitsyn, 2025-08-25:
@wuyuanyi135 , 
The simple examples will be added.

Godoy-Octavio, 2026-09-22:
Hello, @wuyuanyi135 and @lbxhacker 

The new simplified Esp-modbus [examples](https://github.com/espressif/esp-modbus/tree/main/examples/get_started) were added to the project. Please take a look and feel free to give any feedback.