espressif/esp-modbus #134: mb_communication_info_t ignores .ser_opts.mode and defaults to ASCII

i01011001, 2025-08-14:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

often

### Expected behavior

the device should use RTU message format when MB_RTU is specified 

### Actual behavior (suspected bug)

the device uses ASCII message format even though RTU was specified  
 ((checked with usb to tty converter))

### Error logs or terminal output

```shell
E (296640) MB_CONTROLLER_MASTER: mbc_master_get_parameter(131): Master get parameter failure, error=(0x107) (ESP_ERR_TIMEOUT).
```

### Steps to reproduce the behavior

mb_communication_info_t comm = {
    .ser_opts.port = MB_UART_PORT,
    .ser_opts.baudrate = baud_rate,
    .ser_opts.parity = parity,
    .ser_opts.uid = 0,
    .ser_opts.response_tout_ms = 1000,
    .ser_opts.data_bits = UART_DATA_8_BITS,
    .ser_opts.stop_bits = UART_STOP_BITS_1,
    .ser_opts.mode = MB_RTU, // Expected RTU mode
};

### Project release version

v2.1

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

NixOS 

### Shell

ZSH

### Additional context

if .ser_opts.mode is not set, the device defaults to RTU message format

alisitsyn, 2025-08-20:
@i01011001 ,

This is misconfiguration of tcp master issue. Please provide the configuration of the master and ip table or whole project.

alisitsyn, 2025-08-20:
@i01011001 ,

It is clear from your log messages that you changed the example of master and its data dictionary for cid #0, where the UID is changed for your target slave device = 2, but you did not provide the address of slave for this UID in the `slave_ip_address_table`. Please follow the documentation to resolve this.

i01011001, 2025-08-20:
@ali

i01011001, 2025-08-20:
@a

i01011001, 2025-08-20:
@alisitsyn 

One last question, is it possible to run both rtu and tcp master at the same time??

closed 2025-08-25 by i01011001